guysoft/FullPageOS #597: Wifi + page loading issues

COPLGIE, 2024-10-17:
#### What were you doing?

Trying to load up FullPageOS onto a Pi 5 with8gb of RAM.

#### What did you expect to happen?

To load and display multiple webpages after eachother after editing the 2 needed files.

#### What happened instead?

When connected via UTP en editing the fullpageos.txt file with 1 url it works fine.

When i edit fullpagedashboard.txt with multiple url's and fullpageos.txt with http://localhost/fullpagedashboard it only loads the Welcome screen with QR for setup. Does not load contents of fullpagedashboard.

When i try to connect via wifi it seems it does not connect at all for some reason.
Edited the fullpageos_wpa_supplicant.txt file:
## WPA/WPA2 secured
network={
  ssid="SSID which has spaces"
  psk="password"
}

Does nothing. I read that adding "key_mgmt=WPA-PSK" below psk= could solve the issue but nothing.

So i pretty much got 2 issues. 
1. Multiple pages listed in fullpagedashboard.txt is not working
2. Connecting to a wifi network doesnt work. 

#### Was there an error message displayed? What did it say?

Nothing.

#### Version of FullPageOS?

On windows, dont know how i can get into /etc/ but i selected the Nightly from 2024-03-24. ive tried the other 2 versions also, from 2022-03-07 and 2024-03-31 but exactly the same happens.

DietrichsPro, 2024-10-17:
Here's the solution in English:
Step 1. ctrl+alt+F2
Step 2. Login (pi, raspberry)
Step 3. sudo raspi-config
Step 4. Wireless Lan
Step 5. Enter your wifi network details.

DietrichsPro, 2024-10-21:
Mi sugerencia es que habilites el ssh en raspi-config y te conectes con la terminal para modificar la url desde 
sudo nano /boot/firmware/fullpageos.txt

sudo reboot.

cuando recibas la pantalla blanca obliga a chrome a iniciar.
chromium-browser www.paginaweb.com

2Neanderthal, 2024-10-22:
When you are doing the dashboard make sure you have no extra spaces anywhere. I had something similar happen and it happened to be an extra line under my last URL. 

I had to set my wifi using Network manager.  Run "systemctl status NetworkManager" to see if Network manager is running. 
"sudo nano /etc/NetworkManager/system-connections/preconfigured.nmconnection" -This is where I found the network files to be pulling from.

guysoft, 2024-10-22:
Duplicate of #570

There is a NetworkManager way to set the wifi, and there is an even easier one coming up soon.

closed 2024-10-22 by guysoft

COPLGIE, 2024-10-22:
@guysoft unfortunatly, my issue has not been completely resolved. Pages mentioned in the fullpagedashboard.txt do not load.

@2Neanderthal there are no spaces in my fullpagedashboard.txt. I also checked fullpageos.txt but also no spaces.

apartmedia, 2024-10-22:
Is it a duplicate of #534?
Is yes, take a look at [this comment](https://github.com/guysoft/FullPageOS/issues/534#issuecomment-2318374881).
Might be your case.

COPLGIE, 2024-10-23:
@apartmedia Just tried this, and is the solution, it works. Thanks!

Strange tho that it doesnt work with the .txt file while that is how it should work. Perhaps this will get fixed in an update in the future?

apartmedia, 2024-10-23:
A „fix“ would be more related to FullPageDashboard. The way how FullPageOS currently provide the URL‘s (each address in one line of a text file) can not be parsed by FullPageDashboard. The actual FullPageDashboard api can not read multiple line by line URL’s coming from a simple text file. It needs the json format which is currently provided only via the adding function of the tab bar. Or the needed json format would need to be provided directly by FPOS (instead of a text file).