guysoft/FullPageOS #604: missing X server or $DISPLAY

Jojoistcool2509, 2024-11-19:
#### What were you doing?

Trying to get Full Page OS to start. I have a 3,5in GPIO touch screen installed with the drivers configured and the website is supposed to show up on there. During boot it shows the logo on the screen and then it just goes to the terminal.

#### What did you expect to happen?
The browser to start up with the website i set in the Dashboard

#### What happened instead?
Errors

#### Was there an error message displayed? What did it say?
xset: unable to find display ""
feh ERROR: can't open display! It *is* running, yeah?
Failed creating new xdo instance
then some numbers and ERROR:ozone_platform_x11.cc(orm failed to intialize. Exiting.
and also "Missing X server or $DISPLAY"
#### Version of FullPageOS?
the newest nightly build from Raspberry Pi Imager(0.14.0)

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-11-20:
What type of screen display? Does it work on plain Rpi OS?

Jojoistcool2509, 2024-11-20:
> What type of screen display? Does it work on plain Rpi OS?

It's a generic GPIO 3,5inch touch display and yes it works perfectly with Rpi OS when the drivers are installed. If they're not installed the screen is just white. The screen also works in Full Page OS with the LCD-Show drivers installed and will display a what I think os the boot screen with the logo aswell as a terminal that I can type in. Should I try to maybe connect the Pi to a regular HDMI display with the GPIO screen installed to see if maybe something else is displayed on there or if it also displays the terminal and stuff?

guysoft, 2024-11-20:
What are the drivers for this generic display?
I can't tell what is going on without know what actual hardware you have and what it uses. 
For example the official 7inch display works from boot. I am not sure what you actually have.

Jojoistcool2509, 2024-11-20:
I have a Raspberry Pi 3B+, the display says 3.5inch RPi Display, 480x320 Pixel, XPT2046 Touch Controller. The drivers are the LDC35-show from the LCD-show project on GitHub. 

https://github.com/goodtft/LCD-show

The display is rotated by 180 degrees using software, but it already didn't work without the rotation. If you need any log files, tell me which ones and where they're at.

guysoft, 2024-11-20:
It looks like they don't support bookworm, see:
https://github.com/goodtft/LCD-show/issues/350

Also just a few notes from what I can see:
1. The install script of this changes a bunch of things in the image and kernel (I see no LDC35-show, so here is a link to LCD7H-show) :
https://github.com/goodtft/LCD-show/blob/master/LCD7H-show

2. This is a custom script to some hardware. I really do not have the time to support this since I don't have neither the time nor hardware.