guysoft/FullPageOS #608: The hardware on this system lacks support for NEON SIMD extensions. We now require NEON or equivalent architecture extensions on ARM-based machines.

TheExpertNoob, 2024-11-21:
Not an issue per-se, but a fix I found for older hardware. Loaded up the latest Nightly (10-28-2024) on a Pi Zero W (2017) with the Raspberry Pi Imager and eventually got a pop-error mentioned in the title. Reverted back to an older version of chromium-browser with the commands mentioned below.
```
wget https://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-codecs-ffmpeg-extra_104.0.5112.105-rpt2_armhf.deb

wget https://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-browser_104.0.5112.105-rpt2_armhf.deb

sudo apt install ./chromium-codecs-ffmpeg-extra_104.0.5112.105-rpt2_armhf.deb

sudo apt install ./chromium-browser_104.0.5112.105-rpt2_armhf.deb

sudo reboot
```
Hopefully other people that come across this will find this helpful.

marko-pi, 2024-12-04:
I have the same problem, but I get "Connection timed out." when I try to install an old version of Chromium.

By the way, I got this problem automatically when Rasperry Pi OS was updated.

guysoft, 2025-01-02:
Converting to a discussion