guysoft/FullPageOS #616: X11VNC starts and stops continuously

Z3r0shin, 2024-12-07:
#### What were you doing?
Using a Pi3 A+, in portrait mode (commenting out the various dtoverlays in the config.txt file), it shows splash screen, connects to wifi correctly but keeps starting and stopping the X11VNC service continuously until the screens goes blank save from a rapidly blinking underscore in the top left cornerf for about 5 minutes before rebooting with the watchdog.

#### What did you expect to happen?
For it to load the webpage as it did before I've rotated the display

#### What happened instead?
See description above.

#### Was there an error message displayed? What did it say?
It didn't, except for the watchdog timer that rebooted the pi

#### Version of FullPageOS?
0.14.0 nightly

#### Screenshot(s) showing the problem:

[image]
[image]

#### If you are building FullPageOS - provide a build.log that is created for the build

Z3r0shin, 2024-12-07:
I am new to compiling programs and stuff, else I'd remove the X11VNC since I don't need it and add the --no-memcheck flag to the chromium start script to prevent the "under 1g ram" chromium warning from popping up.

Z3r0shin, 2024-12-08:
It seems the X11VNC is dependant on something that the dtoverlay provides and therefore goes exactly as its service file asks it to do : on_failure every 2 seconds.
I have no idea what I am doing.

guysoft, 2024-12-08:
Can you run and share the output of
```
systemctl status x11vnc.service
```
?
It might tell us what the error is.

Z3r0shin, 2024-12-08:
`pi@fullpageos:~ $ systemctl status x11vnc.service
● x11vnc.service - VNC Server for X11
     Loaded: loaded (/etc/systemd/system/x11vnc.service; enabled; preset: enabl>
     Active: active (running) since Sun 2024-12-08 15:30:28 GMT; 897ms ago
   Main PID: 2854
      Tasks: 3 (limit: 667)
        CPU: 1.222s
     CGroup: /system.slice/x11vnc.service

Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 passing arg to lib>
Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 passing arg to lib>
Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 x11vnc version: 0.>
Dec 08 15:30:29 fullpageos x11vnc[2901]: xauth:  unable to generate an authorit>
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: faile>
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: since>
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: faile>
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Main process exited, cod>
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Failed with result 'exit>
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Consumed 1.238s CPU time.
lines 1-18/18 (END)...skipping...
● x11vnc.service - VNC Server for X11
     Loaded: loaded (/etc/systemd/system/x11vnc.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-12-08 15:30:28 GMT; 897ms ago
   Main PID: 2854
      Tasks: 3 (limit: 667)
        CPU: 1.222s
     CGroup: /system.slice/x11vnc.service

Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 passing arg to libvncserver: -rfbport
Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 passing arg to libvncserver: 5900
Dec 08 15:30:28 fullpageos x11vnc[2854]: 08/12/2024 15:30:28 x11vnc version: 0.9.16 lastmod: 2019-01-05  pid: 2854
Dec 08 15:30:29 fullpageos x11vnc[2901]: xauth:  unable to generate an authority file name
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: failed for display=':0'
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: since we are root, retrying with FD_XDM=1
Dec 08 15:30:29 fullpageos x11vnc[2854]: 08/12/2024 15:30:29 -auth guess: failed for display=':0'
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Main process exited, code=exited, status=1/FAILURE
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Failed with result 'exit-code'.
Dec 08 15:30:29 fullpageos systemd[1]: x11vnc.service: Consumed 1.238s CPU time.
`

Seems its something to do with Xauth? It can't find the .Xauthority file?

Z3r0shin, 2024-12-08:
Running xauth list returns
`pi@fullpageos:~ $ xauth list
fullpageos/unix:0  MIT-MAGIC-COOKIE-1  eabf04bbe75459048011cf86d412d5e3
`

Z3r0shin, 2024-12-08:
> pi@fullpageos:/home $ x11vnc -findauth
> XAUTHORITY=

I guess this is not supposed to be blank?

Z3r0shin, 2024-12-13:
Is there anything else I can do? I haven't been able to do anything about it so far...

guysoft, 2024-12-13:
No idea

romankysely, 2025-02-11:
> [@romankysely](https://github.com/romankysely) do you think its worth to switch to it? what binary are you using?

Hi Guy, thx for question and interest. Yes, I do. I have very good experience. I mentioned at closed issue  https://github.com/guysoft/FullPageOS/issues/571  already.  When I use your image I immediatly remoce x11VNC and instlall realVNC instead. Yes, they have Cloud solution paid but if you maintain your list of connections localy, than is it free.

guysoft, 2025-02-24:
@romankysely what commands do you use to install it?
We could add it here:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/start_chroot_script#L121

We could even add an optional switch and have realvnc the default if its indeed better, which I think it has potential to be.

romankysely, 2025-02-24:
Thx. Yes, good idea but I am not programmer. I am afraid to interfere to your code. I am ChatGPT codewriter :-))
Instalation commands:
sudo apt install realvnc-vnc-server # Instalation RealVNC server
sudo systemctl enable vncserver-x11-serviced  # run after boot
sudo systemctl start vncserver-x11-serviced # run realVNC server

Diagnostics:
dpkg -l | grep realvnc # check instalation package realVNC
sudo systemctl status vncserver-x11-serviced  # check running realVNC

Login and password is the same as during install OS puted.