guysoft/FullPageOS #622: Pi5 Audio with TVS99 fails past the first channel

chriskewl, 2025-01-04:
This is regarding the newest nightly for FullPageOS, use this guide below and skip to the section "Getting the Pi Ready":
https://github.com/zshall/program-guide/discussions/66

Make sure you're connected to the internet, it uses streaming elements that will pipe in with music. Some things such as the static on Channel 4 also fails and that's a local file.

Thanks!

guysoft, 2025-01-09:
Hey, I can't understand - what is the issue?

chriskewl, 2025-01-09:
Yes, the audio doesn't work past the first web page. It may be easier if you install it and see for yourself. It should only take 5 minutes from a fresh install. Thanks!

guysoft, 2025-01-16:
I am not sure how to debug this because it does work i the beginning, so its not a configuration issue. Its application-dependent. It might be a bug. in TVS99. It might be a good idea to look though the logs or ask on their issue tracker since they would know better how their application works and why the second channel is not like the first - perhaps they are using something differently.