guysoft/FullPageOS #625: Some web pages do not display

Roberts629, 2025-01-13:
#### What were you doing?

I am using FullPageDashboard to showcase status pages of websites such as Microsoft Office service status (https://portal.office.com/servicestatus). 

#### What did you expect to happen?

Display the page normally similar to Azure status (https://azure.status.microsoft/en-gb/status). 

#### What happened instead?
#### Was there an error message displayed? What did it say?

Chromium displays a blank page with the message "portal.office.com refused to connect"

#### Version of FullPageOS?

Version 0.14.0 (Nightly) on Raspberry Pi 5

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

[image]

I have updated all packages and also tried different versions of FullPageOS with no success. What am I missing?

apartmedia, 2025-01-14:
Seems to be a duplicate of #587
See [this comment](https://github.com/guysoft/FullPageOS/issues/587#issuecomment-2294589799), that might be the reason and explain the functionality. It also mentions some technical limitations and a possible solution if you just need to display a single webpage. If not, note the info about iFrames in general. This is by design and can not be solved really by any chromium landing page. It would need a complete different approach and technique. You should first check if the websites you are planning to load are affected by this limitations. If yes, you can not use the current FPOS solution (or better say not the FP Dashboard solution).

Roberts629, 2025-01-14:
Thank you, that makes more sense now. I will try and find another solution.

closed 2025-01-14 by Roberts629