guysoft/FullPageOS #637: Don't boot on alternative board

SwayLE3, 2025-03-08:
Hey, i just bough an orange pi zero 2w board which is an alternative version of the pi zero 2w.
My project was to use it as a kiosk with Fullpageos, however it doesn't boot. Nothing appear on the screen and the device isn't on my network. I suspect the problem come from the fact that the processor is an Allwin which is the main difference between the official rasp and mine. So i was wondering;

- Is there a way to make that OS boot on my orange pi?

- If no, how can i set it up to work on it via ssh ?because i dont have any keyboard or mouse compatible for the ports on this board.

Thank you !

SwayLE3, 2025-03-08:
> You're right, unfortunately. Every linux image for ARM single board computers has to be prepared for a specific processor family. So there is no Raspberry Pi image that would work with OrangePi.

ok thank you and do you think a build could be asked for that board or is it a too small case  ?

[image]

thank you !

guysoft, 2025-03-08:
@frutabruta its not exactly processor family, since all of these are armf/arm64 arcitecture. The issue they all have a different boot process. Unlike a PC that has a BIOS that manages the boot, embedded single board computers have a bootloader, linux and a file called a "device tree" placed somewhere on the storage media, each device does it a little differently.

@SwayLE3
FullPageOS uses CustomPiOS to build, which does support building for other boards. However I do not have any orange pis, or time to test and maintain another board. You are welcome to build and select the BASE_BOARD setting and see if that works. You will likely need to do a few minor changes to get OrangePi to work.
The board setting is here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/images.yml#L25

See:
https://github.com/guysoft/CustomPiOS/wiki/Building#base-board-selection

SwayLE3, 2025-03-13:
> [@frutabruta](https://github.com/frutabruta) its not exactly processor family, since all of these are armf/arm64 arcitecture. The issue they all have a different boot process. Unlike a PC that has a [...]

Hey, am trying to build the image with docker but i think there is a server-side problem ; 

[image]

Let me know if am doing a mistake or something wrong 
Thanks !

guysoft, 2025-03-13:
You need to login in to docker hub

```
docker login guest
```
Password is guest too.AFAIK