guysoft/FullPageOS #639: Auto login in Ctrl+Alt+F1

frutabruta, 2025-03-25:
Problematic versions: Nightly build from RPI Imager (2024-10-28, 2025-01-28)
Ok version: 2024_03_24
 
When Ctrl+Alt+F1 is pressed, the system skips into console, already logged in. This is a security risk.
In previous versions, log in is required to enter the console - correct behaviour.

RPi imager settings:
- changed password
- changed hostname
- changed locale to cs
- enabled SSH

guysoft, 2025-03-26:
Sounds like a possible regression from here:
https://github.com/RPi-Distro/pi-gen/issues/614#issuecomment-1117558522

guysoft, 2025-03-26:
This patch might be failing:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L98

Can you share the file``/usr/bin/cancel-rename``?
I can pull it myself but I am busy and might not reach it.

frutabruta, 2025-03-26:
I had to add the .txt extension to be able to upload it here.
This one is from the  2025-01-28 version:
[cancel-rename.txt](https://github.com/user-attachments/files/19471060/cancel-rename.txt)

guysoft, 2025-03-30:
That file has:
```bash
else
    if [ -f /var/lib/userconf-pi/autologin ]; then
        SUDO_USER="$1" raspi-config nonint do_boot_behaviour B1
    else
        SUDO_USER="$1" raspi-config nonint do_boot_behaviour B1
    fi
fi
```
Looks like its replacing, it sounds like some reression on rpi's script end but I can't seem to pin it yet

guysoft, 2025-04-29:
XECDesign says in https://github.com/RPi-Distro/pi-gen/issues/614#issuecomment-1117558522
> [@guysoft](https://github.com/guysoft) things seem to work as expected on the PiOS end, AFAICT.
> 
> If the image in question has lightdm with autologin enabled, then I think it would take the 'B4' path.

guysoft, 2025-05-07:
@karlosss I am sorry I have no time to get to this, either use an older version since this is a nightly build. Or help me develop I am afraid. 
The OK version is listed above as:
 version: 2024_03_24

You can still get it here:
https://unofficialpi.org/Distros/FullPageOS/nightly/2024-03-24_2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.zip

The issue was introduced from rpi image version 2024-10-22, since 2024-03-15 does not have it.

guysoft, 2025-08-05:
Re reply here says that currently with how user config script does is written this is not supported :-/ 
Because it can't separate between UI login and tty login.

The issue is in this file:
https://github.com/RPi-Distro/userconf-pi/blob/bookworm/rename-user#L32

And it might be worth to open an issue about it here:
https://github.com/RPi-Distro/userconf-pi/tree/bookworm

guysoft, 2025-08-05:
Posted here too:
https://github.com/RPi-Distro/userconf-pi/issues/9