guysoft/FullPageOS #640: Poor performance of locally hosted site when in fullscreen

pb4ugoout, 2025-03-31:
I'm having a weird issue that I can't track down. I'm self-hosting [this site](https://github.com/Lumon-Industries/Macrodata-Refinement) on an RPi5 running the nightly build from January (bookworm). The site is running in nginx and when browsing to the IP of the Pi, the site runs great. I can open multiple instances without any issues. Granted this probably has more to do with the machine running the javascript than it does the Pi. 

When FullPageOS boots and loads the site, the performance is abysmal. The site slows to a crawl with a second or more of delay between any action. Moving the mouse and trying to drag a box is a chore. Waiting for the numbers to drop into the temper boxes at the bottom is 4-8 seconds. The page never crashes but the performance is awful. 

However, I have found that if I drop into windowed mode (ctrl + f11) the whole issue disappears. The page is snappy and responds as expected. 

The SBC is an 8GB Pi 5 with a cooler on it. The current display is a, 800x480 5" hdmi display. The final product will have a 5" DSI display with capacitive touch. Not sure that is causing the issue though since it works when windowed. I've tried overclocking both the cpu (up to 2.8GHz) and gpu (up to 1GHz) of the pi and it makes no difference. Still runs choppy when fullscreen except now the fan spins up louder to compensate for the added heat. Interestingly, the cpu utilization hovers around 7% when in fullscreen and 31-35% when windowed. This sort of makes sense as it would stand to reason that better performance of the site would come with higher compute utilization. But it also begs the question of why the system is ramping down the compute when the page is displayed fullscreen. 

Open to suggestions and willing to share additional details, including my configs, upon request. Just let me know.

pb4ugoout, 2025-03-31:
For what it's worth, I have the same experience with the Git hosted site on the interwebs. Not sure if it's an issue with the site itself or an issue with how FullPageOS is loading it. ¯\_(ツ)_/¯

guysoft, 2025-04-01:
I have no idea, you might want to ask the chromium team why something like that might happen.
Another guess might be some composite issue, that might be worth asking the rpi foundation kernel team. But Its beyond me.