guysoft/FullPageOS #642: FullpageOS cannot display Uptime-Kuma

masteroftherings, 2025-04-02:
Hi.

I have used FullpageOS on several Raspberry Pi's and a really like the setup. But I have one problem when trying to open Uptime-Kuma in FullpageOS, that won't work. Do you have any idea what can cause this issue? Lack of JavaScript support in Chromium browser or something else? Any suggestions on how to solve this issue are appreciated! 

If I install Raspberry Pi OS with DE, Uptime-Kuma works perfectly well in the Chromium browser so it must be something that is missing in the FullpageOS setup?

Noschvie, 2025-04-02:
> Uptime-Kuma in FullpageOS, that won't work

what exactly do you mean by that?

masteroftherings, 2025-04-02:
Hi, I mean that FullpageOS are unable to display Uptime-Kuma webpage and I would like to solve this somehow.

guysoft, 2025-04-02:
chromium does support javascript. It might be https or webpage policy issues. did you try setting the page directly and without FullPageDashbord? By placing the url in : ``/boot/firmware/fullpageos.txt``? That path is from the readme.

masteroftherings, 2025-04-02:
No, I have only tried to connect to Uptime-Kuma using the FullpageDashboard. But I will try to edit the /boot/firmware/fullpageos.txt before first boot and insert my Uptime-Kuma url here instead. Thanks!

masteroftherings, 2025-04-02:
Thanks for the help! It worked great if I skipped using the FullpageDashboard and edited the fullpageos.txt instead. I wonder what the differences are? It would be nice to be able to use more tabs than just one as the setup is for now.

apartmedia, 2025-04-02:
Most websites do not allow viewing their content inside an iframe which is the technique the Dashboard application is using. The Dashboard needs a complete redesign to be able to offer some kind of multiple tab view. In general this is more a restriction of the target website if you can use iframes or not.

masteroftherings, 2025-04-02:
Thank you for the detailed info!

closed 2025-05-05 by guysoft