guysoft/FullPageOS #646: Building FullPageOS variant for 64bit

alectrocute, 2025-04-15:
I'm using a Raspberry Pi 5 with 8GB of memory. Everything runs better in 64bit but I noticed FullPageOS is 32-only. How can I build with the 64bit version of Raspian?

alectrocute, 2025-04-15:
On my fork, I'm trying to use a 64-bit GitHub runner with a 64-bit Raspian image URL and I'm getting the following build error:

```bash
qemu-arm-static: /usr/bin/bash: Invalid ELF image for this architecture
```

https://github.com/alectrocute/FullPageOS/actions/runs/14478532329/job/40609993848

topa-LE, 2025-05-04:
It is currently completely impossible to build a new FullpageOS with current Debian/Ubuntu distributions.

Unless the developer changes something in the code, the end is unfortunately in sight for FullpageOS ;-(

guysoft, 2025-05-05:
Hey, sorry not to reply sooner. I am pretty busy these days.

Its more than possible to build arm64 builds. I went to length creating CustomPiOS v2 to facilitate that.
See base board section in building:
https://github.com/guysoft/CustomPiOS/wiki/Building#base-board-selection

If you are using hte docker method, you need to set the base board to ``raspberrypiarm64``

```bash
sudo docker exec -it fullpageos-build build -b raspberrypiarm64 -d
```

paspiz85, 2025-11-04:
I see https://github.com/guysoft/CustomPiOS/issues/259 is closed but also https://github.com/guysoft/CustomPiOS/issues/260 is fixed?

However, I'm retrying building FullPageOS for RaspiOS lite x64 with following commands:

```
#!/bin/bash
git clone https://github.com/guysoft/FullPageOS.git
cd FullPageOS/src/image
wget -c --trust-server-names 'https://downloads.raspberrypi.org/raspios_lite_arm64_latest'
cd ..
tee docker-compose.yml >/dev/null <<EOF
services:
  custompios:
    image: ghcr.io/guysoft/custompios:devel
    container_name: fullpageos-build
    tty: true
    restart: always
    privileged: true
    volumes:
      - ./:/distro
    devices:
      - /dev/loop-control
EOF
sudo docker compose up -d
sudo docker exec fullpageos-build bash -c "
    # Detach all loop devices
    losetup -D
    
    # Clean up any leftover mounts
    umount -l /distro/workspace/mount/boot/firmware 2>/dev/null
    umount -l /distro/workspace/mount 2>/dev/null
    
    # Remove old image files
    cd /distro/workspace && rm -f *.img
"
echo "Done! Loop devices cleaned up."
sudo docker exec -it fullpageos-build build
sudo docker compose down
```
but I haven't workspace directory and my log are:
```
pi@berry:~/FullPageOS/src $ sudo docker exec -it fullpageos-build build
+ WORKSPACE_POSTFIX=
+ '[' default '!=' default ']'
+ '[' '' == yes ']'
+ '[' '!' -z '' ']'
+ '[' '' == '' ']'
+ echo 'No BASE_BOARD defined, defaulting to raspberrypiarmhf'
No BASE_BOARD defined, defaulting to raspberrypiarmhf
+ export BASE_BOARD=raspberrypiarmhf
+ BASE_BOARD=raspberrypiarmhf
++ lsof /distro/workspace/mount
++ awk '{print $2}'
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 PID
/usr/bin/build: line 42: kill: PID: arguments must be process or job IDs
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 1
+ for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`
+ kill -9 16
pi@berry:~/FullPageOS/src $ ls
build_dist  config  docker-compose.yml  image  modules  vagrant  variants
```
@guysoft have you any suggestion?

guysoft, 2025-11-04:
Also converting this to a discussion. Its not an issue, because its about an issue.