guysoft/FullPageOS #651: Why does FullPageOS send a heartbeat ping to fullpageos-tracking.gnethomelinux.com every few seconds?

cmenscher, 2025-05-23:
I recently set up a FullPageOS RPi for displaying a network monitor. We have PiHole running locally and have noticed an extremely high number of DNS hits to fullpageos-tracking.gnethomelinux.com. It's every few seconds. ([See this screenshot](https://imgur.com/a/h4XarX9)) We have blocked the domain, but is there a way to disable this tracking ping? Can you explain what it is for?

cmenscher, 2025-05-23:
Hmm never mind I found it in /usr/bin/boot_report. It kept running the curl command because we blocked the domain. I just commented out the code. Sorry, I guess we're opting out of your usage statistics.

FYI, /common.sh also has a command to curl gnethomelinux.com:5000, but the connection is refused when I try. I guess the port isn't open. Not sure that line is ever even run, but I commented that line out as well.

guysoft, 2025-05-23:
Glad you found it. however I am not sure where you see a curl to gnethomelinux.com:5000. That should not exist and I don't recall adding it. Nor can I see it i f I grep both in CustomPiOS or FullPageOS codebase. If you can share a snipplet I'll try to look for that and remove it.

Also of course you can opt out. The usage tracking is really just so I know how much people are using FullPageOS, and what versions are out to see that they get adopted.

I noticed you used "we" there. Just taking the opportunity to note that FullPageOS is 100% free software, I do it in my free time and no one pays me to do it. If you are part of an organization that FullPageOS is helping, please consider making a donation, the link is on the README.

cmenscher, 2025-05-23:
The curl command is on line 194 of common.sh in the function detach_all_loopback. Here's a little context...I can't get github's editor to display the entire script without crazy formatting.

>detach_all_loopback $image_path
>echo "Mounting root parition"
>sudo losetup -f
>curl gnethomelinux.com:5000
>ls /dev

By we I meant my family! I'm going to put the pihole dashboard and maybe ntopweb stuff on a page to show the kids, been teaching my son about networking and we had an extra monitor laying around. But I totally understand the commercial thing and would happily license it if we used it at work. It's great! Thank you for creating it.