guysoft/FullPageOS #656: WebOS

noah-dev-codes, 2025-07-04:
Hey there,
Your guys webOS is great but the thing is I have a made a thing call RasOnTv a media player on Raspberry PI similar to MythTV and I'll be using FullPageOS I'm just wondering where I can get the whole OS like in the SIM it has the files I need that and am I allowed to get the files and upload it to Github and give you guys credits?

guysoft, 2025-07-04:
Hey, I am not sure what you are talking about exactly.

WebOS is some OS by LG which I assume you are not referring to: https://en.wikipedia.org/wiki/WebOS

is RasOnTv that you made open source? Does it have a repo?

What do you mean on the SIM? do you mean on an sd card? Those are two different things:
https://en.wikipedia.org/wiki/SIM_card
https://en.wikipedia.org/wiki/SD_card

FullPageOS is under GPLv3 license, you have to give attribution and release any changes in the code you make under the same license:
https://www.gnu.org/licenses/gpl-3.0.en.html

noah-dev-codes, 2025-07-07:
is RasOnTv that you made open source? Does it have a repo? Yes