guysoft/FullPageOS #660: Docs: dead sample URL in “How to display one URL in full screen” screenshot

snachodog, 2025-07-13:
## Describe the problem
On the official setup guide https://www.otot.tv/fullpageos-setup-instructions/ the section “How to display one URL in full screen” shows a screenshot of fullpageos.txt that contains:
`https://www.thedash.com/dashboard/eevIT7ceU0/embed`

That sample URL now redirects to Spider Strategies homepage.  Anyone who copies it will land on a dead-end page and may wrongly assume their FullPageOS setup or network is at fault. [They have a page elsewhere on the site which explains the service was discontinued and all data deleted on May 1 2019](https://www.spiderstrategies.com/dash/).

## Location in docs

- Page: https://www.otot.tv/fullpageos-setup-instructions/
- Section: “How to display one URL in full screen”
- Image file: [image]

## Steps to reproduce

1. Open the docs page above.
2. Scroll to the “How to display one URL in full screen” section.
3. Point your browser to the URL shown.
4. Visit the URL – it loads to an unexpected page.

## Expected behaviour

Screenshot (and any accompanying example text) should reference a working, neutral placeholder such as:
- `https://example.com`
- `http://localhost/welcome` (which already ships with the image)

or any other live resource you prefer.

## Suggested fix

- Replace the screenshot with one that shows https://example.com (or another valid URL).
- Optionally add a note in the text instructing users to replace the placeholder with their own web-app URL.

Verified 2025-07-13. See docs page source and current screenshot. (References: OTOT setup guide lines 5–6 and image preview.)
Thanks for maintaining FullPageOS!

snachodog, 2025-07-13:
I hope I’ve opened this in the right place. I couldn’t find a separate repository or branch that holds the documentation files for the OTOT setup guide. If the docs are maintained elsewhere (or mirrored in another repo), please let me know and I’ll be happy to re-file the report there instead. Thanks!

guysoft, 2025-07-14:
Cc @tailorvj who maintains that page

closed 2025-10-18 by guysoft