guysoft/FullPageOS #675: Not Rotating

igotostem, 2025-09-25:
Raspberry PI5 with 0.14.0

Trying to get the screen to rotate to portrait / 90 degrees. I saw other threads about adding display_hdmi_rotate=1, that has not worked.

Below is config file
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Rotate HDMI screen
# 1 is 90 degrees. 2 is 180 degress. 3 is 270 degrees.
display_hdmi_rotate=3

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
# dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

#  [cm5]
#dtoverlay=dwc2,dr_mode=host

[all]
# enable raspicam
start_x=1
gpu_mem=256
disable_splash=1

```

Any suggestions on what to do?

apartmedia, 2025-09-26:
Did you try the official recommendations from the [FAQ](https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position)?

igotostem, 2025-09-26:
> Did you try the official recommendations from the [FAQ](https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position)?

Yes, lcd_rotate did not work. display_hdmi_rotate also did not work, neither did those lines of code

Those were all recommendations for raspberry pi 4 and older, not sure if the two hdmi ports have something to do with it

apartmedia, 2025-09-26:
And you also tried setting the variable in start_gui script as mentioned in the FAQ?
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/opt/scripts/start_gui#L7
I installed FPOS a while ago (an earlier version) via RPi imager on a Pi4 and could flip the screen and touch by using this setting (to be honest I used 180° horizontal, not 90° vertical). It‘s worth a try. The main thing in my case was not only flipping the screen, but also the touch. I was using the official pi touch screen. So it might be a differente Story depending on your screen model and the used drivers.